Annex-Engineering/Sherpa_Mini-Extruder #23: Teflon tube attachement

Lefuneste83, 2021-03-27:
Hi there. I have been using the Sherpa Mini on my Voron V0 and it is a great design. The only issue I have (and it is a constant pain) is that the bowden tube does not hold in place in the hexagonal hole at the top of the extruder. There is no way I could use glue on a teflon tube. It would not hold to the material. Have you considered a slight modification of this hole to allow for a bowden adapter such as a PC4-M5 or a small clip version. I will try to drill out this hex hole to fit such an adapter but it would be great to provision something for it in the base design.

ChipCE, 2021-03-27:
I agree with @Lefuneste83. Or provide the cad file,it's very hard to deal with the stl with too many curve corners. I can only modify the file in tinkle cad, other softwares I tried failed to map the faces into solid object.

szaghi, 2021-03-27:
Hi all, I have just worked at simple mod using a collet to retain the PTFE tube at the top of the extruder, it is really wip, some modifications are needed (the collet hole is too much small, the collet is too much tight in the current version), I hope to fix it soon, but I can share my simple mod if someone is interested (see pictures attached). It is only a temporary mod until an "official" version with integrated PTFE connector will be available.

[image]

szaghi, 2021-03-28:
Hi all,

I have fixed some tolerances of my stupid collet-mod, I attach this version below. 

Cheers

[image]
[sherpa-collet-ptfe.zip](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/files/6217217/sherpa-collet-ptfe.zip)

Annex-Engineering, 2021-03-28:
You guys are welcome to put in a pull request for a user mod. There's a folder on the top level of the directory for this.

Annex-Engineering, 2021-03-30:
user mod has been  added, closing issue

closed 2021-03-30 by Annex-Engineering

szaghi, 2021-03-30:
Hi all,

I have just created a PR (https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/pull/24) with a *user mod* to provide a version of the housing core with an integrated PTFE retaining collet (see the picture below).

Cheers

[image]

drewgwallace, 2024-05-12:
I am not finding this adaptation for a PTFE retainer in the user mods - was it removed?

https://github.com/Annex-Engineering/Annex-Engineering_User_Mods/tree/main/Extruders/Sherpa_Mini/Extruder_Mods