Annex-Engineering/Sherpa_Mini-Extruder #25: collet adapter version has too large hole

5PAZT1C, 2021-04-24:
the hole for the collet adapter as measured is 6.5-6.7mm, for proper fitment with e3d collets the hole needs to be 6mm

5PAZT1C, 2021-04-24:
to further expand on the issue, I have one collet that 'fits' however it is very easy to pop out and the raised lip is also too short, so that even with the retaining clip it wiggles quite a bit. Suggest changing diameter of lip to 6.00mm from it's current ~6.5mm as printed, and increasing the length of the lip by 1.5mm to decrease the amount of play in the collet fitting once retaining clip is inserted. 

Alternatively, suggest providing a part number for a collet to be used with this mod, preferably one available from multiple vendors

szaghi, 2021-04-25:
@5PAZT1C 

Thank you very much for the insight, I dimensioned the mod with the collet I had (an unknown-manufacturer Chinese one) and with the tolerance of my printer. Now that the CAD file of version 1 has been provided I can re-desing the collet adapter with the dimensions you pointet out. ASAP I will provide the step of my mod with fixed dimensions.

Regards,
Stefano

5PAZT1C, 2021-04-25:
if it helps I found the datasheet for the e3d collet:
[image]
[image]

I would think that making both the inner diameter of the bore and the lip slightly undersized with a taper leading from the inner diameter to the lip would result in a good fitting collet adapter. You can see the documentation for the e3d v6 lite here to see what I mean:

https://wiki.e3d-online.com/images/8/84/DRAWING-LITE6-175-SINK.png

szaghi, 2021-04-25:
@5PAZT1C 

This helps a lot, I will use this E3D datasheet as a reference, thank you again.

5PAZT1C, 2021-04-25:
> @5PAZT1C
> 
> This helps a lot, I will use this E3D datasheet as a reference, thank you again.

sure thing! Glad I could help contribute, even if it's something small. I'll keep an eye out for the updated version and do a test print, I'll follow up on this once printed

szaghi, 2021-04-25:
> > @5PAZT1C
> > 
> > This helps a lot, I will use this E3D datasheet as a reference, thank you again.
> 
> sure thing! Glad I could help contribute, even if it's something small. I'll keep an eye out for the updated version and do a test print, I'll follow up on this once printed

I hope to find the time tomorrow

5PAZT1C, 2021-04-25:
> > > @5PAZT1C
> > > This helps a lot, I will use this E3D datasheet as a reference, thank you again.
> > 
> > 
> > sure thing! Glad I could help contribute, even if it's something small. I'll keep an eye out for the updated version and do a test print, I'll follow up on this once printed
> 
> I hope to find the time tomorrow

no rush, I have to buy more ABS before I can reprint it anyways lol

Annex-Engineering, 2021-06-19:
seems to be a dead issue, closing for now

closed 2021-06-19 by Annex-Engineering

szaghi, 2021-06-19:
@Annex-Engineering 

Yes, I am sorry, but currently I have not the time to fix it, I hope to find it soon

szaghi, 2021-09-08:
[sherpa-ptfe-collet-retain.zip](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/files/7131332/sherpa-ptfe-collet-retain.zip)

@5PAZT1C 

I am very sorry for my delay, but I was very busy. I have attached a preliminary version of the modified housing with PTFE collet retainer integrated (the whole step and the STL of the modified housing): I designed the collet and housing-collet-retainer using the references you have indicated. Can you give me a feedback (no rush, when you can/want) about it before creating a new PR?

My best regards

5PAZT1C, 2021-09-08:
thanks! I will print it and check fitment when I have time, however it will be quite a while before I can reprint it and check fitment, as I am leaving the country for vacation for a month tommorrow afternoon

szaghi, 2021-09-09:
@5PAZT1C 

Do not worry, no rush, enjoy your vacation