Annex-Engineering/Sherpa_Mini-Extruder #36: Can you make this part more reliable?

pasha4ur, 2023-09-24:
Hello

I printed this part from ABS+ on different printers. It constantly breaks down. I have already replaced 5 parts and 2 sets of gears.

Make it more reliable. Please.

[https://i.ibb.co/XVK9r0Z/photo-2023-09-24-14-00-50.jpg](https://i.ibb.co/XVK9r0Z/photo-2023-09-24-14-00-50.jpg)

anlin93, 2023-09-24:
White ABS is known to be prone to all sorts of problems and is not recommended for structural parts. We very rarely see failures around the idler arm pivot when printed out of ASA (or ABS in almost any other colour). The failure in your case seems to be related to inadequate layer adhesion which can be caused by a number of things, one of them being the material. Overextrusion or badly printed overhangs can also stress the plastic around the pivot, so if the pivot shaft requires more than a firm push against a hard surface to insert, you may need to adjust your print settings and/or ream out the pivot hole. There is not enough detail about the print settings, material or the printer to provide a more comprehensive answer. My suggestion is to try a different material (black ASA preferred) and make sure the print settings are appropriate for the material and your printer. If you require further assistance, feel free to post in #sherpa_help_forum on our Discord server, which is our main support channel.

closed 2023-09-24 by anlin93

pasha4ur, 2023-09-24:
I printed it in black too. This is bad design problem.

dan-and, 2023-09-24:
I'm sorry, but I haven't had any problems with the structure. I'm using 5 mini and micros and they never broke. I'm not even using the recommended materials, but an ASA from a spanish manufacturer.

anlin93, 2023-09-24:
Sherpa has been out 3 years and so far there have only been a handful of reports of this issue, basically all of which can be attributed to bad prints. As the problem remains isolated to your case, no design changes are planned. If you want to post your modified file as a mod, feel free to make a pull request. Any further communication should happen on our Discord server, as instructed above. I will add a note to the main repo README that GitHub issues are not a forum for technical support.