Annex-Engineering/Sherpa_Mini-Extruder #40: Release 2 CAD missing idler_arm_long

310weber, 2024-01-01:
The CAD for release 2 seems to be missing the idler_arm_long part (or rather the part shows up in the model tree but is empty).

Steps to reproduce:
1. download parasolid model [sherpa_mini_release_2.x_t](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/blob/master/CAD/sherpa_mini_release_2.x_t) from commit [fa76f44](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/commit/fa76f449e86c1e4ee2c25fc557be01597a096812)
2. import model to Onshape
3. navigate to "idler_arm_long^housing_sherpa_mini_a3_r16(FDM)" in the model tree
4. part is empty

[image]

ChaosBlades, 2024-02-28:
It is not missing... just really really small 🤣 I compared Release 1 and Release 2 arms. The geometry is different so you can't simply split the old thumb lever and add it to the new idler arm short.
[image]
Also, would be nice when this is fixed it is updated to a Step file instead of the Parasolid file. The ONLY free way a user can use that file is add it to Onshape then export it as a Step. Who knows how long it will be before that is locked behind a pay wall as well.

So people don't have to make an Onshape account to use the current CAD I have already converted it.
[sherpa_mini_release_2.zip](https://github.com/Annex-Engineering/Sherpa_Mini-Extruder/files/14431012/sherpa_mini_release_2.zip)

anlin93, 2024-02-28:
This looks to be an issue related to Onshape. The X_T opens fine in SolidWorks, long idler arm included
[image]

closed 2024-02-28 by anlin93